bytedeco/procamcalib #1: Troubleshoouting using high definition camera

Gershgorine, 2015-02-19:
Hi,
First of all, i’d like to thank you for your work, i’ve been searching an easy way to calibrate a projector and a camera in order to correct the projection distortion for a long time.  I’m trying to get a 3D map of an object using the Moiré Fringe analysis, for that I’m using a IDS camera with a high resolution (3840*2748), in order to get enough precision in my measurement.
The problem is that I can’t run the calibration with this camera, I thinks it’s due to the too high definition of each image. 
I try many different settings, without any success.  I build the source files, but I’m not good enough in JAVA and I’m getting lost. 
So my question is: Is there a way to use high definition camera in the settings or in the code that I missed?

Thank for your help in advance
Grégoire

saudet, 2015-02-19:
There shouldn't be any limitations in that respect. Are you saying that you tried with lower resolutions and it's working fine?

In any case, could you also provide the error message that you get when you try? Thanks!

Gershgorine, 2015-02-20:
Hi, 
yes it worked perfectly with a web cam, on my computer.
When i'm defining a lower number of pixel in height and width ( like 1024*1024) with the ids camera it's working as well. There is a live on the screen.
But when i'm launching the program in full resolution, it's trying to put the live on screen, but it crashes after a short time. The memory used is around 0.8GB.
I'm running the latest  version of the procamcalib program.
My drivers are all uptodate, and i think every thing is in 64 byte.

Thanks a lot for your help.

---------------  T H R E A D  ---------------

Current thread (0x00000000586c6800):  JavaThread "pool-4-thread-1" [_thread_in_native, id=5692, stack(0x0000000066220000,0x0000000066320000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x000000006c8a1000

Registers:
RAX=0xffffffffffffc3f1, RBX=0x000000007e750040, RCX=0x000000006c8e1000, RDX=0x000000000000c3f1
RSP=0x000000006631eea0, RBP=0x000000006c8b0040, RSI=0x000000000000fd21, RDI=0x0000000000000ab3
R8 =0x0000000000004fae, R9 =0x0000000075777d51, R10=0x000000009b60fa42, R11=0x0000000000000eff
R12=0x0000000098030040, R13=0x0000000056e4f920, R14=0x0000000000000000, R15=0x0000000000000000
RIP=0x000007feee27803f, EFLAGS=0x0000000000010286

Top of Stack: (sp=0x000000006631eea0)
0x000000006631eea0:   0000000063da8290 0000000000000000
0x000000006631eeb0:   00000000586c6800 0000000056e4f920
0x000000006631eec0:   0000000000000000 00000eff00000000
0x000000006631eed0:   0000000075777d51 0000000000000f00
0x000000006631eee0:   00000ab300000000 00000ab400000780
0x000000006631eef0:   000000000000055a 000000007ee60040
0x000000006631ef00:   000000009a200040 0000000097c20040
0x000000006631ef10:   00000000580072a0 0000000000000000
0x000000006631ef20:   0000000074d70050 0000000000000000
0x000000006631ef30:   0000000000000000 000000006631f1b0
0x000000006631ef40:   0000000000000080 000007feee278575
0x000000006631ef50:   0000000000000001 0000000056e4f920
0x000000006631ef60:   0000000000000080 000000006631f070
0x000000006631ef70:   000000006631f010 000000006631f008
0x000000006631ef80:   000000006631f000 000000006631f018
0x000000006631ef90:   0000000056e4f920 000007feee26506e 

Instructions: (pc=0x000007feee27803f)
0x000007feee27801f:   33 ff 48 8b cd 41 8b d0 83 fe 01 7c 23 0f 1f 40
0x000007feee27802f:   00 48 63 01 3b c2 75 08 41 8b c0 41 ff c0 eb 04
0x000007feee27803f:   8b 44 85 fc ff c2 89 01 48 83 c1 04 3b d6 7e e1
0x000007feee27804f:   4c 8b b4 24 c8 00 00 00 41 8d 48 ff 41 89 8d 40 

Register to memory mapping:

RAX=0xffffffffffffc3f1 is an unknown value
RBX=0x000000007e750040 is an unknown value
RCX=0x000000006c8e1000 is an unknown value
RDX=0x000000000000c3f1 is an unknown value
RSP=0x000000006631eea0 is pointing into the stack for thread: 0x00000000586c6800
RBP=0x000000006c8b0040 is an unknown value
RSI=0x000000000000fd21 is an unknown value
RDI=0x0000000000000ab3 is an unknown value
R8 =0x0000000000004fae is an unknown value
R9 =0x0000000075777d51 is an unknown value
R10=0x000000009b60fa42 is an unknown value
R11=0x0000000000000eff is an unknown value
R12=0x0000000098030040 is an unknown value
R13=0x0000000056e4f920 is an unknown value
R14=0x0000000000000000 is an unknown value
R15=0x0000000000000000 is an unknown value

Stack: [0x0000000066220000,0x0000000066320000],  sp=0x000000006631eea0,  free space=1019k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jniARToolKitPlus.dll+0x3803f]
C  [jniARToolKitPlus.dll+0x38575]
C  [jniARToolKitPlus.dll+0x2506e]
C  [jniARToolKitPlus.dll+0xc0ec]
C  0x00000000023d3874

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  org.bytedeco.javacpp.ARToolKitPlus$Tracker.arDetectMarkerLite(Lorg/bytedeco/javacpp/BytePointer;ILorg/bytedeco/javacpp/ARToolKitPlus$ARMarkerInfo;Lorg/bytedeco/javacpp/IntPointer;)I+0
j  org.bytedeco.javacv.MarkerDetector.detect(Lorg/bytedeco/javacpp/opencv_core$IplImage;Z)[Lorg/bytedeco/javacv/Marker;+234
j  org.bytedeco.javacv.ProCamGeometricCalibrator$3.run()V+42
j  java.util.concurrent.Executors$RunnableAdapter.call()Ljava/lang/Object;+4
j  java.util.concurrent.FutureTask.run()V+42
j  java.util.concurrent.ThreadPoolExecutor.runWorker(Ljava/util/concurrent/ThreadPoolExecutor$Worker;)V+95
j  java.util.concurrent.ThreadPoolExecutor$Worker.run()V+5
j  java.lang.Thread.run()V+11
v  ~StubRoutines::call_stub

---------------  P R O C E S S  ---------------

M Arguments:
java_command: C:\Users\Gregoire  PdeL\Documents\Calib_projo_NEW\procamcalib-bin\\modules\procamcalib.jar
java_class_path (initial): C:\Users\Gregoire  PdeL\Documents\Calib_projo_NEW\procamcalib-bin\\modules\procamcalib.jar
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\java\jre1.8.0_31\bin
PATH=C:\Program Files\java\jre1.8.0_31\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\EgisTec MyWinLocker\x86;C:\Program Files (x86)\EgisTec MyWinLocker\x64;;C:\OpenCV\build\x86\vc10\bin;C:\Program Files\Apache Software Foundation\apache-maven-3.2.5\bin;.;
USERNAME=Gregoire  PdeL
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 37 Stepping 5, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 7 , 64 bit Build 7601 Service Pack 1

CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 37 stepping 5, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, ht, tsc, tscinvbit

Memory: 4k page, physical 3857108k(1397056k free), swap 7712360k(4098420k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.31-b07) for windows-amd64 JRE (1.8.0_31-b13), built on Dec 17 2014 21:00:28 by "java_re" with MS VC++ 10.0 (VS2010)

time: Fri Feb 20 11:27:49 2015
elapsed time: 44 seconds (0d 0h 0m 44s)

Gershgorine, 2015-02-20:
Hi,
yes it worked perfectly with a web cam, on my computer.
When i'm defining a lower number of pixel in height and width ( like 1024*1024) with the ids camera it's working as well. There is a live on the screen.
But when i'm launching the program in full resolution, it's trying to put the live on screen, but it crashes after a short time. The memory used is around 0.8GB.
I'm running the latest version of the procamcalib program.
My drivers are all uptodate, and i think every thing is in 64 byte.

Thanks a lot for your help.

 A fatal error has been detected by the Java Runtime Environment:

  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000007feee27803f, pid=1948, tid=5692

 JRE version: Java(TM) SE Runtime Environment (8.0_31-b13) (build 1.8.0_31-b13)
 Java VM: Java HotSpot(TM) 64-Bit Server VM (25.31-b07 mixed mode windows-amd64 compressed oops)
 Problematic frame:
C  [jniARToolKitPlus.dll+0x3803f]

Failed to write core dump. Minidumps are not enabled by default on client versions of Windows

 If you would like to submit a bug report, please visit:
   http://bugreport.java.com/bugreport/crash.jsp
 The crash happened outside the Java Virtual Machine in native code.
 See problematic frame for where to report the bug.
---------------  T H R E A D  ---------------

Current thread (0x00000000586c6800):  JavaThread "pool-4-thread-1" [_thread_in_native, id=5692, stack(0x0000000066220000,0x0000000066320000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x000000006c8a1000

Registers:
RAX=0xffffffffffffc3f1, RBX=0x000000007e750040, RCX=0x000000006c8e1000, RDX=0x000000000000c3f1
RSP=0x000000006631eea0, RBP=0x000000006c8b0040, RSI=0x000000000000fd21, RDI=0x0000000000000ab3
R8 =0x0000000000004fae, R9 =0x0000000075777d51, R10=0x000000009b60fa42, R11=0x0000000000000eff
R12=0x0000000098030040, R13=0x0000000056e4f920, R14=0x0000000000000000, R15=0x0000000000000000
RIP=0x000007feee27803f, EFLAGS=0x0000000000010286

Top of Stack: (sp=0x000000006631eea0)
0x000000006631eea0:   0000000063da8290 0000000000000000
0x000000006631eeb0:   00000000586c6800 0000000056e4f920
0x000000006631eec0:   0000000000000000 00000eff00000000
0x000000006631eed0:   0000000075777d51 0000000000000f00
0x000000006631eee0:   00000ab300000000 00000ab400000780
0x000000006631eef0:   000000000000055a 000000007ee60040
0x000000006631ef00:   000000009a200040 0000000097c20040
0x000000006631ef10:   00000000580072a0 0000000000000000
0x000000006631ef20:   0000000074d70050 0000000000000000
0x000000006631ef30:   0000000000000000 000000006631f1b0
0x000000006631ef40:   0000000000000080 000007feee278575
0x000000006631ef50:   0000000000000001 0000000056e4f920
0x000000006631ef60:   0000000000000080 000000006631f070
0x000000006631ef70:   000000006631f010 000000006631f008
0x000000006631ef80:   000000006631f000 000000006631f018
0x000000006631ef90:   0000000056e4f920 000007feee26506e 

Instructions: (pc=0x000007feee27803f)
0x000007feee27801f:   33 ff 48 8b cd 41 8b d0 83 fe 01 7c 23 0f 1f 40
0x000007feee27802f:   00 48 63 01 3b c2 75 08 41 8b c0 41 ff c0 eb 04
0x000007feee27803f:   8b 44 85 fc ff c2 89 01 48 83 c1 04 3b d6 7e e1
0x000007feee27804f:   4c 8b b4 24 c8 00 00 00 41 8d 48 ff 41 89 8d 40 

Register to memory mapping:

RAX=0xffffffffffffc3f1 is an unknown value
RBX=0x000000007e750040 is an unknown value
RCX=0x000000006c8e1000 is an unknown value
RDX=0x000000000000c3f1 is an unknown value
RSP=0x000000006631eea0 is pointing into the stack for thread: 0x00000000586c6800
RBP=0x000000006c8b0040 is an unknown value
RSI=0x000000000000fd21 is an unknown value
[... 30 lines cut ...]

M Arguments:
java_command: C:\Users\Gregoire  PdeL\Documents\Calib_projo_NEW\procamcalib-bin\\modules\procamcalib.jar
java_class_path (initial): C:\Users\Gregoire  PdeL\Documents\Calib_projo_NEW\procamcalib-bin\\modules\procamcalib.jar
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\java\jre1.8.0_31\bin
PATH=C:\Program Files\java\jre1.8.0_31\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\EgisTec MyWinLocker\x86;C:\Program Files (x86)\EgisTec MyWinLocker\x64;;C:\OpenCV\build\x86\vc10\bin;C:\Program Files\Apache Software Foundation\apache-maven-3.2.5\bin;.;
USERNAME=Gregoire  PdeL
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 37 Stepping 5, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 7 , 64 bit Build 7601 Service Pack 1

CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 37 stepping 5, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, ht, tsc, tscinvbit

Memory: 4k page, physical 3857108k(1397056k free), swap 7712360k(4098420k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.31-b07) for windows-amd64 JRE (1.8.0_31-b13), built on Dec 17 2014 21:00:28 by "java_re" with MS VC++ 10.0 (VS2010)

time: Fri Feb 20 11:27:49 2015
elapsed time: 44 seconds (0d 0h 0m 44s)

saudet, 2015-02-20:
I see, it's crashing in ARToolKitPlus, which isn't known to be very stable. To debug this, it would help to know at which line exactly it crashes...

Gershgorine, 2015-02-20:
How could i get it? 

here is all the error file. THanks again for your help.
 A fatal error has been detected by the Java Runtime Environment:

  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000007feee27803f, pid=1948, tid=5692

 JRE version: Java(TM) SE Runtime Environment (8.0_31-b13) (build 1.8.0_31-b13)
 Java VM: Java HotSpot(TM) 64-Bit Server VM (25.31-b07 mixed mode windows-amd64 compressed oops)
 Problematic frame:
 C  [jniARToolKitPlus.dll+0x3803f]

 Failed to write core dump. Minidumps are not enabled by default on client versions of Windows

 If you would like to submit a bug report, please visit:
   http://bugreport.java.com/bugreport/crash.jsp
 The crash happened outside the Java Virtual Machine in native code.
 See problematic frame for where to report the bug.

---------------  T H R E A D  ---------------

Current thread (0x00000000586c6800):  JavaThread "pool-4-thread-1" [_thread_in_native, id=5692, stack(0x0000000066220000,0x0000000066320000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x000000006c8a1000

Registers:
RAX=0xffffffffffffc3f1, RBX=0x000000007e750040, RCX=0x000000006c8e1000, RDX=0x000000000000c3f1
RSP=0x000000006631eea0, RBP=0x000000006c8b0040, RSI=0x000000000000fd21, RDI=0x0000000000000ab3
R8 =0x0000000000004fae, R9 =0x0000000075777d51, R10=0x000000009b60fa42, R11=0x0000000000000eff
R12=0x0000000098030040, R13=0x0000000056e4f920, R14=0x0000000000000000, R15=0x0000000000000000
RIP=0x000007feee27803f, EFLAGS=0x0000000000010286

Top of Stack: (sp=0x000000006631eea0)
0x000000006631eea0:   0000000063da8290 0000000000000000
0x000000006631eeb0:   00000000586c6800 0000000056e4f920
0x000000006631eec0:   0000000000000000 00000eff00000000
0x000000006631eed0:   0000000075777d51 0000000000000f00
0x000000006631eee0:   00000ab300000000 00000ab400000780
0x000000006631eef0:   000000000000055a 000000007ee60040
0x000000006631ef00:   000000009a200040 0000000097c20040
0x000000006631ef10:   00000000580072a0 0000000000000000
0x000000006631ef20:   0000000074d70050 0000000000000000
0x000000006631ef30:   0000000000000000 000000006631f1b0
0x000000006631ef40:   0000000000000080 000007feee278575
0x000000006631ef50:   0000000000000001 0000000056e4f920
0x000000006631ef60:   0000000000000080 000000006631f070
0x000000006631ef70:   000000006631f010 000000006631f008
0x000000006631ef80:   000000006631f000 000000006631f018
0x000000006631ef90:   0000000056e4f920 000007feee26506e 

Instructions: (pc=0x000007feee27803f)
0x000007feee27801f:   33 ff 48 8b cd 41 8b d0 83 fe 01 7c 23 0f 1f 40
0x000007feee27802f:   00 48 63 01 3b c2 75 08 41 8b c0 41 ff c0 eb 04
0x000007feee27803f:   8b 44 85 fc ff c2 89 01 48 83 c1 04 3b d6 7e e1
0x000007feee27804f:   4c 8b b4 24 c8 00 00 00 41 8d 48 ff 41 89 8d 40 

Register to memory mapping:

RAX=0xffffffffffffc3f1 is an unknown value
RBX=0x000000007e750040 is an unknown value
RCX=0x000000006c8e1000 is an unknown value
RDX=0x000000000000c3f1 is an unknown value
RSP=0x000000006631eea0 is pointing into the stack for thread: 0x00000000586c6800
RBP=0x000000006c8b0040 is an unknown value
RSI=0x000000000000fd21 is an unknown value
RDI=0x0000000000000ab3 is an unknown value
R8 =0x0000000000004fae is an unknown value
R9 =0x0000000075777d51 is an unknown value
R10=0x000000009b60fa42 is an unknown value
R11=0x0000000000000eff is an unknown value
[... 372 lines cut ...]

VM Arguments:
java_command: C:\Users\Gregoire  PdeL\Documents\Calib_projo_NEW\procamcalib-bin\\modules\procamcalib.jar
java_class_path (initial): C:\Users\Gregoire  PdeL\Documents\Calib_projo_NEW\procamcalib-bin\\modules\procamcalib.jar
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\java\jre1.8.0_31\bin
PATH=C:\Program Files\java\jre1.8.0_31\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\EgisTec MyWinLocker\x86;C:\Program Files (x86)\EgisTec MyWinLocker\x64;;C:\OpenCV\build\x86\vc10\bin;C:\Program Files\MVTec\HALCON-11.0\bin\x64-win64;C:\Program Files\MVTec\HALCON-11.0\FLEXlm\x64-win64;C:\Program Files\Apache Software Foundation\apache-maven-3.2.5\bin;.;
USERNAME=Gregoire  PdeL
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 37 Stepping 5, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 7 , 64 bit Build 7601 Service Pack 1

CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 37 stepping 5, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, ht, tsc, tscinvbit

Memory: 4k page, physical 3857108k(1397056k free), swap 7712360k(4098420k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.31-b07) for windows-amd64 JRE (1.8.0_31-b13), built on Dec 17 2014 21:00:28 by "java_re" with MS VC++ 10.0 (VS2010)

time: Fri Feb 20 11:27:49 2015
elapsed time: 44 seconds (0d 0h 0m 44s)

saudet, 2015-02-24:
We have to debug in C++. Could you prepare a sample video file that we could load with FFmpeg, so I could reproduce the issue here? Thanks!

poqudrof, 2017-11-13:
Is this bug still open @saudet  ? 
I proposed a fix on ARToolkitplus a few years ago on a similar issue. 

https://bugs.launchpad.net/artoolkitplus/+bug/1498585

The fix was included in JavaCPP-presets.

saudet, 2017-11-13:
I haven't looked back at it recently. Maybe it's fixed, yes. Thanks for the info!

closed 2018-02-27 by saudet