bytedeco/procamcalib #2: Trouble on Running Script

yangyanzhe, 2015-08-04:
Hi, 

Thanks for your working. I encounter some problems and hope to get help. Does anyone else encounter the same problem?

I work under Ubuntu 14.04 with JAVA 1.8.0_51(Oracle). The binary is downloaded from the link in readme.md. When I execute ./procamcalib-oceanlook or ./procamcalib-nativelook, it will crash. The traceroute is shown below:

SEVERE: Could not start ProCamCalib
java.lang.Exception: java.lang.NoClassDefFoundError: Could not initialize class org.bytedeco.javacpp.opencv_core
	at org.bytedeco.procamcalib.MainFrame.<init>(MainFrame.java:204)
	at org.bytedeco.procamcalib.MainFrame$25.run(MainFrame.java:1309)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.NoClassDefFoundError: Could not initialize class org.bytedeco.javacpp.opencv_core
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at org.bytedeco.javacpp.Loader.load(Loader.java:390)
	at org.bytedeco.javacpp.Loader.load(Loader.java:358)
	at org.bytedeco.javacpp.opencv_core$CvErrorCallback.<clinit>(opencv_core.java:8233)
	at org.bytedeco.procamcalib.MainFrame.<init>(MainFrame.java:186)
	... 15 more

Look forward to your reply.

Thanks!

yangyanzhe, 2015-08-04:
I change the platform tobe Windows 7. No problem.

saudet, 2015-08-05:
This means your version of Linux is using old version of the system libraries:
https://github.com/bytedeco/javacpp-presets/issues/22
Please try to rebuild from source.

saudet, 2015-08-06:
FYI, some are available here:
https://github.com/poqudrof/javacpp-presets/releases/tag/1.0

yangyanzhe, 2015-08-19:
Thanks for your reply. :+1:

saudet, 2015-10-30:
This should be fixed in version 1.1. Thanks for reporting!

closed 2015-10-30 by saudet