bytedeco/procamcalib #4: Using procamcalib with CCDbcamera

DanLighter, 2016-04-19:
Hi Samuel,

Firstly, thank you for developing this awesome piece of software!

I am basically using it to calibrate the positions of cameras and projectors for a surface capture imaging system. This consists of 1 webcam, 1 projector and 1 CCD camera. 

I have managed to get the calibration procedure to work when using just the webcam and projector straight through OpenCV setting. I am now trying to add the additional CCD camera simultaneously.

I have downloaded and compiled the libdc1394 library and when I run it I can get the camera to capture frames.

However what settings would I need to point the procamcalib software in the direction of the CCD camera please? When I run at the moment I get the following error. ps. I am not very familiar with Java.

[image]

These are the settings I currently have (basically default) for the ProCamCalib software.

[image]

Many thanks in advance for your help!

Dan

saudet, 2016-04-19:
You've compiled libdc1394 for Windows and you've confirmed that it works for your camera?

saudet, 2016-04-19:
I see you did. So, we would need to generate JNI bindings for that now... Could you try to follow the instructions on the following page?
https://github.com/bytedeco/javacpp-presets/#build-instructions
Or at least could you tell me how you got libdc1394 to work on Windows, and I'll add it to the cppbuild.sh script? Thanks!

DanLighter, 2016-04-19:
Hi sam, thanks for the quick response. I compiled libdc1394 using MinGW to create an exe file which can run the camera. So I have all the cpp files available in the environment path. I will try following those build instructions and let you know how i get on! Thanks

saudet, 2016-04-20:
You didn't need to specify any special configure flags? It just worked?

DanLighter, 2016-04-20:
The only thing I changed from default settings in the cpp code was the image resolution to 1024 x 1024. When I compiled it, it recognized the camera from the IEEE1394 port.

saudet, 2016-04-21:
It requires the CMU 1394 Camera Driver to be installed though:
```
checking for GetCmdrVersion in -l1394camera... no
configure: error: lib1394camera not found or too old. Please upgrade to 6.4.5 or a more recent version.
```
And we might want to install `libusb` as well:
`pacman -S mingw-w64-x86_64-libusb mingw-w64-i686-libusb`

I can easily produce an EXE with that, but not a DLL like we need for Java. Do you know how to make it create a DLL?

saudet, 2016-04-21:
Oh, I think I got something working. Get the latest commit above and try a `mvn clean install` on that!

DanLighter, 2016-04-21:
Yes sorry I forgot to mention I installed the CMU driver and changed the driver on the camera to the CMU one. 
Ok looks great. I am away from work for the next 9 days on a conference, but I will try the above when I am back on my uni computer with the CCD. Thanks for the help!

DanLighter, 2016-05-12:
hi samuel, I have been back and having a play with this again this week. we have been trying desperately to get it to compile but had a lot of issues along the way

am i right in saying that I first need to build javacpp in maven and javacpp presets then once that is in the repository run install on the libdc1394 build? 
is it possible to simply compile the libdc1394 in maven and the copy over the jar files into the existing folder. at the moment i am just running 
mvn install --projects .,libdc1394

or will i need to compile all the different plugins by building and running mvn clean install on all of these files after downloading them all?

would it also be possible to get a copy of the compiled jar file that worked for you please so i can test it in the software to see if it will be compatible with my CMU?
thanks for your help in advance

saudet, 2016-05-13:
You could build libdc1394 only yes, but you would need to backport the changes to JavaCPP 1.1.

Anyway, we should be making a new release by next week, so you could also wait a bit.

saudet, 2016-05-16:
I've released version 1.2 which should work out of the box with libdc1394 on Windows.

Thanks for reporting this!

closed 2016-05-16 by saudet

DanLighter, 2016-05-16:
Hi Samuel,

Thanks for the update. I have been playing with it this morning. At the moment I am having difficulty with the new 1.2 version. It crashes on my system the moment I do anything ie. try to load in previous calibration pcc settiongs. I get the following error

java platform se binary has stopped working

And then it closes. However version 1.1 seems to work so I wouldn't think it is an issue with my java platform but I could be wrong. I have the most up to date version. I have also tried it on multiple computers and I get the same issue.

Thanks for your help again,

saudet, 2016-05-16:
Well, it works here (Windows 10), so I will not be able to debug something that works fine.

If you could try to build from source and figure out what makes it work on your system, that would help.

DanLighter, 2016-05-16:
Ok I will have another go at the build again and let you know how I get on

saudet, 2016-05-16:
Thanks! BTW, instructions for that are here:
https://github.com/bytedeco/javacpp-presets/wiki/Build-Environments#windows-x86-and-x86_64

saudet, 2016-05-17:
You know, I don't have a camera to test libdc1394 at the moment, so there are good chances this is related to the new libdc1394 module for Windows. If you remove the libdc1394*jar files from the modules directory, does it still crash? If not, then we could limit our investigation to that module...

DanLighter, 2016-05-17:
yep we did have a go at just removing those files, but it still crashed. the good news is before it crashes when i have the camera plugged in it recognises it. do you know of a way i could debug or get a problem signature when running the module to give you a bit more info? I tried running the file through command line but got no information back on why it crashed. 

i am just trying the build again from source and getting further than before

saudet, 2016-05-17:
When the JVM crashes, it usually creates a log file with a name like `hs_err_xxxx.log` either in the current directory or in a temporary directory somewhere. That should contain useful information.

DanLighter, 2016-05-17:
Ok I think i have found the Java error. Does this mean anything useful? (sorry for the lack of knowledge, i am not a computer science student)

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000076e5a12f, pid=5852, tid=3280
#
# JRE version: 7.0_07-b10
# Java VM: Java HotSpot(TM) 64-Bit Server VM (23.3-b01 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [ntdll.dll+0x5a12f]  RtlFreeHeap+0x3f
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x000000000cd78800):  JavaThread "AWT-EventQueue-0" [_thread_in_native, id=3280, stack(0x000000000def0000,0x000000000dff0000)]

siginfo: ExceptionCode=0xc0000005, reading address 0xffffffffffffffff

Registers:
RAX=0x00000000002c7380, RBX=0x900000ac45f35480, RCX=0x0000000000420000, RDX=0x0000000000000000
RSP=0x000000000dfec510, RBP=0x900000ac45f35490, RSI=0x0000000000000000, RDI=0x0000000000000000
R8 =0x900000ac45f35490, R9 =0x0000000000000158, R10=0x00000000c0000005, R11=0x000000000dfec748
R12=0x0000000000420000, R13=0x0000000000000000, R14=0x000000000dfec770, R15=0x000000000cd78800
RIP=0x0000000076e5a12f, EFLAGS=0x0000000000010246

Top of Stack: (sp=0x000000000dfec510)
0x000000000dfec510:   0000000000000000 000000000dfec530
0x000000000dfec520:   0000000000000001 0000000076e1eb4c
0x000000000dfec530:   0000000000420000 0000000000000000
0x000000000dfec540:   0000000000000000 00000000000000b0
0x000000000dfec550:   00000000111c0100 000007feec2f7e08
0x000000000dfec560:   0000000000000000 0000000000000000
0x000000000dfec570:   900000ac45f35490 0000000000420000
0x000000000dfec580:   0000000000000001 000007feec2f7e5f
0x000000000dfec590:   0000000000420000 0000000000000001
0x000000000dfec5a0:   900000ac45f35490 0000000000000000
0x000000000dfec5b0:   0000000000000000 0000000076c01a0a
0x000000000dfec5c0:   000000000dfec770 000000009cf98dd8
0x000000000dfec5d0:   0000000000000000 000000000dfec6f0
0x000000000dfec5e0:   0000000000000000 0000000076c01a0a
0x000000000dfec5f0:   00000000111bf160 00000000674d7167
0x000000000dfec600:   0000000000000000 000000009cf98dd8 

Instructions: (pc=0x0000000076e5a12f)
0x0000000076e5a10f:   00 00 0f ba 61 74 18 0f 82 1c 02 00 00 f6 41 78
0x0000000076e5a11f:   01 75 60 41 f6 c0 0f 75 3f 49 8d 58 f0 0f 0d 0b
0x0000000076e5a12f:   80 7b 0f 05 75 0b 0f b6 43 0e 48 c1 e0 04 48 2b
0x0000000076e5a13f:   d8 f6 43 0f 3f 75 47 45 33 c9 48 8b d1 4c 8b c3 

Register to memory mapping:

RAX=0x00000000002c7380 is an unknown value
RBX=0x900000ac45f35480 is an unknown value
RCX=0x0000000000420000 is an unknown value
RDX=0x0000000000000000 is an unknown value
RSP=0x000000000dfec510 is pointing into the stack for thread: 0x000000000cd78800
RBP=0x900000ac45f35490 is an unknown value
RSI=0x0000000000000000 is an unknown value
RDI=0x0000000000000000 is an unknown value
R8 =0x900000ac45f35490 is an unknown value
R9 =0x0000000000000158 is an unknown value
R10=0x00000000c0000005 is an unallocated location in the heap
[... 319 lines cut ...]
0x000007fefa4c0000 - 0x000007fefa5e5000 	C:\Windows\system32\dbghelp.dll

VM Arguments:
java_command: C:\Users\dml413.CCA-0932\Downloads\procamcalib-1.2-bin\procamcalib-bin\\modules\procamcalib.jar --laf javax.swing.plaf.metal.MetalLookAndFeel
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\Java\jdk7
PATH=C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Perl\site\bin;C:\Perl\bin;C:\Program Files (x86)\MiKTeX 2.9\miktex\bin;C:\Perl64\site\bin;C:\Perl64\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\SSH Communications Security\SSH Secure Shell;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;c:\Program Files (x86)\Microsoft SQL Server\100\Tools\Binn\;c:\Program Files\Microsoft SQL Server\100\Tools\Binn\;c:\Program Files\Microsoft SQL Server\100\DTS\Binn\;C:\Matlab\R2009a\bin;C:\Matlab\R2009a\bin\win32;C:\Program Files (x86)\TortoiseSVN\bin;c:\Program Files (x86)\Microsoft SQL Server\90\Tools\binn\;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\PROGRA~2\IVIFOU~1\VISA\WinNT\Bin;C:\Program Files\Thorlabs\Scientific Imaging\bin;C:\Program Files (x86)\Common Files\Pleora;C:\Program Files\Common Files\Pleora;C:\MinGW\bin;C:\apache-maven-3.3.9\bin
USERNAME=dml413
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 23 Stepping 10, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 7 , 64 bit Build 7601 Service Pack 1

CPU:total 4 (4 cores per cpu, 1 threads per core) family 6 model 23 stepping 10, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, tsc

Memory: 4k page, physical 6184800k(4112344k free), swap 12367744k(9046528k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (23.3-b01) for windows-amd64 JRE (1.7.0_07-b10), built on Aug 28 2012 18:28:27 by "java_re" with unknown MS VC++:1600

time: Mon May 16 11:21:19 2016
elapsed time: 47 seconds

DanLighter, 2016-05-17:
Ok I think i have found the Java error. Does this mean anything useful? (sorry for the lack of knowledge, i am not a computer science student)
[hs_err_pid5852.txt](https://github.com/bytedeco/procamcalib/files/268175/hs_err_pid5852.txt)

saudet, 2016-05-17:
I've seen crashes like that with older system libraries on Linux before. Maybe the same bug is in the current version of libraries on Windows in MSYS2...

DanLighter, 2016-05-17:
one question, if i boot in linux and run procamcalib either 1.1 or 1.2 i shouldnt have the same problem?

saudet, 2016-05-17:
It's possible, but I think it's a bug in some versions of the libraries that's used on both platforms...

saudet, 2016-05-21:
Since this problem seems to be happening only during deallocation, you could try to set the "org.bytedeco.javacpp.nopointergc" system property to "true". This will lead to memory leaks, but the calibration session doesn't usually take too much time, so I don't think it would be an issue. :)

saudet, 2016-12-08:
While doing some testing for the release of version 1.3, I encountered those crashes and setting the "org.bytedeco.javacpp.nopointergc" property to "true" does indeed work around the problem so I have added that flag to the launch script by default...