bytedeco/procamcalib #11: Error getting static method ID of org/bytedeco/javacpp/Loader/putMemberOffset

shine636363, 2016-09-13:
Hi,

Could someone kindly help me with the below issues? One mentioned  that my Linux version is using old version of the system libraries. I also tried other compiled files from https://github.com/poqudrof/javacpp-presets/releases/tag/1.0. However, it is still not working.

java version "1.7.0_79"
OpenJDK Runtime Environment (IcedTea 2.5.5) (7u79-2.5.5-0ubuntu0.14.04.2)
OpenJDK 64-Bit Server VM (build 24.79-b02, mixed mode)
Sep 13, 2016 4:35:14 PM org.bytedeco.procamcalib.MainFrame$25 run
SEVERE: Could not start ProCamCalib
java.lang.Exception: java.lang.NoClassDefFoundError: Could not initialize class org.bytedeco.javacpp.opencv_core
	at org.bytedeco.procamcalib.MainFrame.<init>(MainFrame.java:204)
	at org.bytedeco.procamcalib.MainFrame$25.run(MainFrame.java:1309)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:312)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:745)
	at java.awt.EventQueue.access$300(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:706)
	at java.awt.EventQueue$3.run(EventQueue.java:704)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:715)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)
Caused by: java.lang.NoClassDefFoundError: Could not initialize class org.bytedeco.javacpp.opencv_core
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:274)
	at org.bytedeco.javacpp.Loader.load(Loader.java:464)
	at org.bytedeco.javacpp.Loader.load(Loader.java:409)
	at org.bytedeco.javacpp.opencv_core$CvErrorCallback.<clinit>(opencv_core.java:8855)
	at org.bytedeco.procamcalib.MainFrame.<init>(MainFrame.java:186)
	... 15 more

If I need to rebuild the whole thing, could you please also provide some hints for this. Thanks again.

saudet, 2016-09-13:
Errors about `putMemberOffset` means that you are using an older version of JavaCPP with newer binaries, so use the latest version of JavaCPP.

closed 2016-09-13 by saudet

shine636363, 2016-09-14:
Hi,

I download the binary from search.maven.org/remotecontent?filepath=org/bytedeco/procamcalib/1.2/procamcalib-1.2-bin.zip

How could I find the right version for this?

saudet, 2016-09-14:
That's fine, but you seem to be trying to use with other things that need
version 1.0: That won't work.