bytedeco/procamcalib #12: Opencv grabber problem

SimoneErcoli, 2017-10-13:
I am trying to use your procamcalib software (Mac os X build). I have some problem with the framegrabber. The application starts and the camera image "flash" but after 30/40 seconds I have this error       

java.lang.RuntimeException: Unknown exception.
at org.bytedeco.javacpp.opencv_videoio$VideoCapture.read(Native Method)

at org.bytedeco.javacv.OpenCVFrameGrabber.trigger(OpenCVFrameGrabber.java:232)

I am using opencv grabber because I have a "load class error" trying to use the other grabbers

saudet, 2017-10-14:
Yes, OpenCV doesn't work very well on Mac OS X. FFmpeg works better usually:
https://trac.ffmpeg.org/wiki/Capture/Webcam
According to that site, if we set the format to "avfoundation" and specify an empty string or "default" for the device filename or path it should work.

SimoneErcoli, 2017-10-16:
I have to recompile the project ?

saudet, 2017-10-16:
No, it should work with the current set of binaries just fine. Choose "FFmpeg" as your "FrameGrabber".

SimoneErcoli, 2017-10-16:
what about deviceFile and devicePath ???

saudet, 2017-10-16:
See the comment above.

SimoneErcoli, 2017-10-16:
[image]

how to use this information to fill the camera path ??

saudet, 2017-10-16:
Well, according to https://trac.ffmpeg.org/wiki/Capture/Webcam#AVFoundation
we can use the device number as "filename"...

SimoneErcoli, 2017-10-16:
ok...using terminal works but i don't understand where I have to put this value in your interface

saudet, 2017-10-16:
Like you said, in "devicePath"...

SimoneErcoli, 2017-10-16:
[image]

same error also with the name of the camera

saudet, 2017-10-16:
What did you use for the "format" field?

SimoneErcoli, 2017-10-16:
nothing

SimoneErcoli, 2017-10-16:
solved ;-) ...I have just to find the right dimension and frame rate...but now I am able to open the webcam e try to match the two boards

saudet, 2017-10-16:
Good! Setting the format to "avfoundation" as I mentioned above would probably help as well.

closed 2017-10-24 by saudet