bytedeco/procamcalib #13: RealSense F200 enable_stream error

xsorifc28, 2017-11-10:
I receive the following error when running the calibrator with a RealSense F200 camera.

`failed to fetch unpacker, most likely because enable_stream was not called`

I've built the RealSense drivers for Ubuntu 14.04 using [the provided instructions](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md)  and the examples work fine.

I appreciate any insight you might have regarding this.

[image]

saudet, 2017-11-10:
@poqudrof any ideas?

poqudrof, 2017-11-13:
Hello, 
I am not sure how well the librealsense works with procamcalib.
Anyway, you don't really need the librealsense drivers to work with procamcalib. 

1. Plug the camera.
2. Install the librealsense drivers (optionnal). 
You can get the color feed using the linux UVC drivers. You can check if it works with the program `guvcview`. With FFMEPG use `/dev/video0` as device, OpenCV use `0`.
3. Use the FFMPEG or OpenCV drivers to load the realsense camera. 
4. Start the calibration. 

If you need to calibrate with IR feeds, we can either try to patch the software, or you can calibrate with the color camera then use calibration from the device. The Realsense cameras provide extrinsic calibration for each camera (color - IR - IR2 - depth). 

Please tell me if you have another use / request. 
Cheers, 
Jeremy.

xsorifc28, 2017-11-20:
I was able to fix the issue by referencing [this pr](https://github.com/bytedeco/javacv/pull/677). Seems like later in the development this fix was removed.

saudet, 2017-11-20:
The last release of ProCamCalib was last year, so it doesn't contain that fix yet. Replacing `modules/javacv.jar` with the one from JavaCV 1.3.3 should work though. In any case, glad to hear it works! Thanks for the feedback.

saudet, 2018-01-17:
Version 1.4 has been released and includes the fix. Enjoy!

closed 2018-01-17 by saudet

saudet, 2018-02-16:
So, this wasn't fixed in latest release of JavaCV? Could you send a PR to reapply the fix? Thanks!