bytedeco/procamcalib #15: Ran into out of memory error

andrewsunst, 2018-01-30:
I used the flycapture2 camera on windows 10 64-bit and when it was running it seemed the video shown frame by frame with about 10 sec interval. The projector only projected white background and at the end it showed "could not perform calibration java.lang.OutOfMemoryError:Java heap space." How do I solve this problem?

saudet, 2018-01-30:
Try to reduce the frame rate of your camera to 1 FPS if possible.

andrewsunst, 2018-01-31:
The memory problem is getting better but the projector is still not projecting the pattern. Sometimes I can see the patterns on my screen but not the projector. I have set the projector right(because it is a different size) in the projector settings. Is there something else I can do to fix this? Now it only projects white background.

saudet, 2018-01-31:
Have you tried Java 9? It did fix a lot of display issues like that on new
OSs...

andrewsunst, 2018-02-01:
Does it actually work with java 9? It throws NoClassDefFoundError every time.

saudet, 2018-02-01:
Oh, you're right. NetBeans 8.x doesn't support Java 9. We'll need to update
that...

saudet, 2018-02-01:
You could also try to use OpenGL:
https://github.com/bytedeco/procamcalib/blob/master/procamcalib-oceanlook.cmd
It might work better with your display driver.

andrewsunst, 2018-02-01:
I tried the setting with opengl but I don't think it saves the whole program. It ran successfully once today and it was projecting the patterns on the board however when I was changing the light to make the pattern recognizable in the camera it crashed and couldn't even start calibration anymore.

andrewsunst, 2018-02-01:
In the bug report says "Failed to write core dump. Minidumps are not enabled by default on client versions of Windows". Which didn't happen a lot before.

saudet, 2018-02-02:
Might help to disable both OpenGL and DirectX then:
https://docs.oracle.com/javase/8/docs/technotes/guides/2d/flags.html

andrewsunst, 2018-02-02:
I have set the opengj and d3d to false which do make the program running but it unable recognize the patterns on the board.

andrewsunst, 2018-02-02:
Oh, I am using my laptop and now it runs good without so many wired problems. However, the  projected pattern is often unrecognizable and it keeps calibrating until memory full and crushes. I havn't calibrate it successfully once. Will you be able to provide the environment parameters (like the light intensity of surrounding) and what should be show in the GUI when calibrating(should there be anything shown in the right side of the GUI when calibrating or it stay blank until calibration finishes)?

saudet, 2018-02-02:
You'll need to figure out how to make the markers visible with your camera. There's nothing magic about it. You can try to change the settings in MarkerPatterns, MarkerDetector, and GeometricCalibrator. You should also try to change the settings of your camera and of your projector, but in the end, if the markers are not visible, calibration is not possible. What should happen during calibration is stated clearly in the README.md file:
> When you want ProCamCalib to take an image for calibration, keep the board as steady as possible for a few seconds until you see the camera image "flash". Please refer to my Procams 2009 paper and the demo video to understand further how to perform calibration. Also, please note that ProCamCalib may fail to detect markers properly if they are not clear or big enough. Please adjust properly the focus, exposure, resolution, etc. of your projectors and cameras.

andrewsunst, 2018-02-06:
I made it run one time and I am wondering if I can use the realsense camera on windows to compare the ground truth. I saw the description of the realsense is F200 and when I was trying my realsense vf0800 and it couldn't work with it. So will the program actually support my camera?

saudet, 2018-02-06:
I don't believe so, we'll probably need to add support for librealsense2 for that I think. @poqudrof?

poqudrof, 2018-02-09:
I really don't know, I see you updated librealsense to 1.12.1, so I guess it compiles ? 

I had such a camera (SR300), but did not try it on windows, however it worked fine and the preset was written for it.  Hum, I have an idea to try it on windows. I create an AR library with many video inputs, most of them come from javacpp-presets. It will be announced in the coming weeks / months. 

I try it out with my library on windows and I will document the procedure if it works !

poqudrof, 2018-02-09:
The release from JavaCPP-presets works on windows ! 
:smile:

saudet, 2018-02-11:
@poqudrof It would be great if we could integrate your code to access more cameras to JavaCV!!

poqudrof, 2018-02-12:
Most of my video inputs come from JavaCV. I guess the issue is the integration JavaCV - procamcalib that I don't master yet and I just did not take the time to get a deeper look into it. 
For my test, I just plugged the camera on Windows 10 64bits, downloaded the full jar of javacpp-presets and started the camera. My test code will be released soon as a [PapARt update](https://github.com/poqudrof/PapARt).

saudet, 2018-02-12:
Sounds good, but where is the support 
for Orbbec coming from, for example? Would it make sense to create a FrameGrabber around that to be able to use it with ProCamCalib?

poqudrof, 2018-02-13:
Yes, I had a look at it yesterday. I could translate the code to a FrameGrabber. Orbbec provides their support with OpenNI and they provide the jar and binaries precompiled. 

They use an Apache licence, so I guess we could integrate it and distribute it :+1:

saudet, 2018-02-13:
Ok, cool, send a pull request when you have the time and let's look at that! Thanks

andrewsunst, 2018-02-15:
I found the issue [https://github.com/bytedeco/procamcalib/issues/13](url) and I am now having the same problem with the realsense F200. So now I am trying to compile javacv 1.3.3 myself and I met some problem with the dependency. I downloaded the current release of the javacpp and added to the javacv project but things like javacpp.RealSense and javacpp.opencv seems not included in the javacpp.jar file. Would you help with this issue? 
Thank you.

saudet, 2018-02-15:
That issue should be fixed with version 1.4. Are you still having issues with version 1.4?

andrewsunst, 2018-02-16:
Yes, I first used procamcalib 1.4 and it had this problem so I replaced the javacv in it with the javacv 1.4 just in case the javacv in procamcalib is not updated and it still had this problem.

andrewsunst, 2018-02-19:
I added all the deleted lines about enable_stream in [here](https://github.com/bytedeco/javacv/pull/677/files)  and it started working well with realsense cameras(I used F200 and SR300). However, after I take enough pictures it throws "java.lang.exception Image grabbed from Camera 0 is null, unexpected end of stream" which is in the CalibrationWorker class. Would you help me with this problem?
Thank you.

saudet, 2018-02-19:
Thanks for checking this out! Could you please also send a pull request to
have those lines added to JavaCV?

andrewsunst, 2018-02-21:
The pull request is sent but the code has the bug which I mentioned before and I hope maybe you could help with it.

saudet, 2018-02-21:
Yes, thank you! @poqudrof could you take a look at this?

closed 2018-02-25 by andrewsunst