bytedeco/procamcalib #16: procamlib doesn't run

teganscott, 2018-02-07:
When trying to run "procamlib-nativelook" I get this error: java.lang.Exception: java.lang.NoClassDefFoundError: javax/activation/DataContentHandlerFactory

Running on OSX. 
Can the PS3 eye camera only run on Windows for this calibration?

Thanks for your help,
Tegan

saudet, 2018-02-07:
It should work, yes, but Java 9 isn't supported. Please try with Java 8.

teganscott, 2018-02-08:
Thank you, I ended up getting it to work on Windows, but will try this for the future

I've been trying for the past 6 hours to calibrate the system. I've only seen it flash twice in that time to capture an image. It finds both the projected and printed markers but rarely ever settles on an image. This was on a logitech webcam, now I am using the PS3 Eye and am having similar problems. If you have any suggestions it would be much appreciated.

saudet, 2018-02-08:
Nothing in particular other than adjusting the settings of ProCamCalib, your camera, and of your projector, as per the README.md file:
> When you want ProCamCalib to take an image for calibration, keep the board as steady as possible for a few seconds until you see the camera image "flash". Please refer to my Procams 2009 paper and the demo video to understand further how to perform calibration. Also, please note that ProCamCalib may fail to detect markers properly if they are not clear or big enough. Please adjust properly the focus, exposure, resolution, etc. of your projectors and cameras.

saudet, 2018-02-08:
Marking as duplicate of issue #15.

closed 2018-02-08 by saudet