bytedeco/procamcalib #17: Test Calibration

teganscott, 2018-02-08:
Is there a method I can use to test the calibration? When I put the calibration numbers into my own software (which works with opencv and calibrations from other systems) it doesn't coorrelate the camera and projector correctly. 

Thank you,
Tegan

saudet, 2018-02-08:
Could you copy/paste typical results that you get here?

teganscott, 2018-02-08:
Below is what procamlib gives me, and the formatting I have for my calibration files. These two will be slightly different because the projector and camera are in different positions. The biggest difference is in the extrinsic rotation of my old calibrations was a 3x1 matrix, and the new one is 3x3. In my code I have the following that converts the rotation matrix into Rodrigues: 

cv::Vec3d axis_angle;
        double data[9] = {9.9981890125721906e-01, -8.7052645857853533e-03,
            1.6922856062103579e-02, 1.6082024959920501e-02,
            8.6194239428941977e-01, -5.0675110004795354e-01,
            -1.0175124667329560e-02, 5.0693148185441583e-01,
            8.6192629588781566e-01};
        cv::Mat origRot = cv::Mat(3,3, CV_64F, data);
        cv::Rodrigues(origRot, axis_angle);
        cv::Mat rot  = cv::Mat(3, 1, CV_64F, axis_angle);

This is what procamlib gives me:

%YAML:1.0
---
Cameras:
   - Camera  0
Projectors:
   - Projector  0
Camera  0:
   imageWidth: 640
   imageHeight: 480
   responseGamma: 0.
   cameraMatrix: !!opencv-matrix
      rows: 3
      cols: 3
      dt: d
      data: [ 7.7581750624364702e+02, 0., 3.2712277369620023e+02, 0.,
          7.7610945139991838e+02, 2.3484626839931556e+02, 0., 0., 1. ]
   distortionCoeffs: !!opencv-matrix
      rows: 1
      cols: 4
      dt: d
      data: [ -1.1150538574733834e-01, 1.7528485231520349e-01,
          1.3777229721069281e-04, 3.5512804807152055e-03 ]
   extrParams: !!opencv-matrix
      rows: 10
      cols: 6
      dt: d
      data: [ -2.4156499206165061e-01, 1.8012775548542140e-01,
          3.0971126521233794e+00, 9.7672612756320257e+02,
          8.4309337872134495e+02, 3.0579462522394706e+03,
          -7.8549242710261080e-01, -2.8501749151166361e-01,
          -2.9269460537869385e+00, 8.6784058799846707e+02,
          8.7237706917774949e+02, 2.5111457941924082e+03,
          1.7173840480802446e-01, 7.6806524901152817e-01,
          -3.0011869667530067e+00, 8.1736575306125781e+02,
          5.9774869098573743e+02, 3.1628262239991532e+03,
          -4.9985867968159597e-02, -9.3919224795251288e-01,
          -2.8871041470444330e+00, 1.1068227286276594e+03,
          1.0505865212093518e+03, 3.0583236718773715e+03,
          1.2579745239764833e-01, 9.2067055083533952e-02,
          -3.1256843279217237e+00, 9.0416180531444070e+02,
          7.0449061666552848e+02, 2.5043720925455700e+03,
          2.7428722866650290e-01, 8.7245690431954001e-02,
          3.1078726288431393e+00, 6.8503074658642504e+02,
          1.0020846281253456e+03, 4.1669747122728004e+03,
          -1.1503040335019099e+00, 1.7024453055326852e-01,
          2.8854476605390929e+00, 4.5169684320598648e+01,
          1.1418006977246384e+03, 5.0062320468696498e+03,
          -1.2036048151279497e+00, 9.0617781989513636e-04,
          -2.7793536264733043e+00, 4.9855943180165428e+02,
          9.8652834329722725e+02, 3.5770730312324645e+03,
          3.9387072333177947e-01, -1.3150287815934847e-01,
          -3.0678627249891148e+00, 3.5969986263255259e+02,
          1.1399110179951804e+03, 5.0285896568295348e+03,
          3.1649409187408778e-01, -1.4772840685344468e-01,
          -3.0460147144952132e+00, 1.3429765374386300e+03,
          1.1414031663775518e+03, 3.4867023648561662e+03 ]
[... 192 lines cut ...]
sensorSize_width: 0
sensorSize_height: 0
distCoeffs: !!opencv-matrix
   rows: 5
   cols: 1
   dt: d
   data: [ -9.8245651394649747e-02, 5.6098143869503347e-01,
       1.1589744437146346e-02, 4.8286095453437001e-03,
       5.6681736350551748e-01 ]
reprojectionError: 5.2733808755874634e-01

**EXTRINSICS:**
%YAML:1.0
Rotation_Vector: !!opencv-matrix
   rows: 3
   cols: 1
   dt: d
   data: [ 6.8210825021344387e-02, 2.8757581667304671e-02,
       -3.0531880320480636e-02 ]
Translation_Vector: !!opencv-matrix
   rows: 3
   cols: 1
   dt: d
   data: [ 2.4228539839093036e-01, 3.1220677070266731e+00,
       2.1248321280799489e+00 ]

saudet, 2018-02-08:
Good! The errors don't look too big, so the calibration is OK. You'll just
need to plug in the numbers correctly, yes.