bytedeco/procamcalib #20: Instruction about the experiment setting

andrewsunst, 2018-09-11:
I contacted with you about a few months before and now I have done the calibration and used the result for reconstruction. However, the error of your method just doesn't feel right…… I used 3*4 as the size of those tags but the error of your method was like ten times of ours so I am wondering if you can give some instructions about under what conditions your method will perform better?

saudet, 2018-09-11:
So just 12 tags? That's not enough, you'll need a lot more than that to get
anything accurate...

saudet, 2018-09-19:
For more details about how I obtained good results, please read this paper:
http://www.ok.ctrl.titech.ac.jp/res/PCS/publications/procams2009.pdf
If you have any questions left after reading it, please let me know! Thanks

closed 2018-09-19 by saudet