bytedeco/procamcalib #21: Unable to project or get Camera settings

nikhil4595, 2019-03-21:
Hi @saudet 

I have connected a DLP Lightcrafter and Imaging Source Machine Camera to perform calibration of the system. Unfortunately, everytime I try to run the calibration with entered parameters, the Camera 0  and the projector 0 window shows nothing(plain white and black respectively).
Is there anything that I might be missing or haven't configured correctly. Thank you so much.

Regards,
Nikhil

saudet, 2019-03-21:
What kind of parameters? Does it work with a webcam? Also, if you try to use the monitor as output, do you see the pattern displayed?

nikhil4595, 2019-03-21:
parameters like frameRate, image height,width. It does not work with a webcam either(also running with OpenCv as frame grabber) . I tried using the monitor as output , still getting a black screen on the Projector window

saudet, 2019-03-21:
What do you mean by "not working" with a webcam?

nikhil4595, 2019-03-21:
I figured that out, my system webcamera was defect, and not recognized even with the device manager. So the camera works if I disable it. Thanks a lot.
Is there any solution for the resolution issue , because 640x480 does not cover the whole scene in the case of my setup and the markers are not detected for 1024*768 or higher resolutions.

saudet, 2019-03-21:
You can try to set the resolution with the "imageWidth" and "imageHeight" parameters.

nikhil4595, 2019-03-21:
I already tried that , bot for VideoInput and OpenCV with 1024*768 and 1920*1080. 
A couple of times it logged complete error file, but mostly , every other time it flips the stream/image both horizontally and vertically.

saudet, 2019-03-21:
If you can't change it that way, it probably means there is some other way usually some tool that you need to use to change the resolution of your camera.

closed 2019-04-21 by saudet