bytedeco/procamcalib #22: Using calibration data with OpenCV

zekeadam, 2019-04-02:
Hello,

I'm a bit confused about how to use the resulting values with OpenCV and cant find too much information on the topic. Could you give me some pointers about how to align the cameras image the projectors image using OpenCV? I was experimenting with cv.undistort() and a few other functions but can't figure it out.

Thanks in advance, great software!

saudet, 2019-04-02:
There is code that does that as part of JavaCV and ProCamTracker, check it
out!

saudet, 2019-04-03:
If you have any questions about them, please feel free to ask here.

zekeadam, 2019-04-04:
Thank you for the help!
Do it get it right that the `undistort(double[] xd, double[] k)` function undistorts the cameras image then the `distort(double[] xu, double[] k)` (in  javacv/ProjectiveDevice.java) distorts it so its aligned to the projector?

saudet, 2019-04-04:
Yes, that's the idea, but those methods are for individual points.

closed 2019-04-21 by saudet