bytedeco/procamcalib #24: procamcalib doesn't run when calibration one camera and a projector

Sousouam, 2019-06-03:
I try to calibrate a single camera with a projector but calibration doesn't run. 
I am using windows 10 and java 1.8. 
When i run the command procamcalib-nativelook.cmd, I get as following :

C:\Users\AAAA\Desktop\procamcalib-bin>procamcalib-nativelook.cmd
java version "1.8.0_191"
Java(TM) SE Runtime Environment (build 1.8.0_191-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.191-b12, mixed mode)

the chessboard image still in full screen and nothing else. 
the calibration still running without any result ! 

help please !

saudet, 2019-06-03:
You'll need to print the "board" pattern and wave it in front of the
projector and camera.

Sousouam, 2019-06-03:
thank you for your replay, i did it for several seconds but nothing was happened. 
How much time calibration required to be achieved ?

saudet, 2019-06-03:
Enough markers need to be detected for the calibration to proceed. Adjust
your projector and camera settings to capture images that are as clear as
possible.