bytedeco/procamcalib #26: Realsense2 device is found, but cannot run program

hyeonjang, 2022-02-11:
OS: ubuntu 20.04
Camera: Realsense2 D435i, L515

Hello,
When I tried to run this program, I encounter this message.

[image]

Also, on a console the message indicates 
`
Realsense2 devices found: 1
`

I cannot figure out where this problem occurs.

Could you help  me?
Thanks

saudet, 2022-02-11:
@cansik Any ideas?

cansik, 2022-02-11:
For the `RealSense2FrameGrabber` it is necessary to first enable the necessary streams. Here an example ([samples/RealSense2DepthMeasuring.java#L44-L59](https://github.com/bytedeco/javacv/blob/master/samples/RealSense2DepthMeasuring.java#L44-L59)):

```java
// enable the color & depth stream of the realsense camera
rs2.enableColorStream(640, 480, 30);
rs2.enableDepthStream(640, 480, 30);

rs2.start();
```

I assume `procamcalib` needs the `start` method to directly enable at least the color input. In my opinion this should be included into the `RealSense2FrameGrabber`, if no stream has been activated, instead of the exception. It is possible to activate de default color stream by just passing `0` as parameter for width, height and framerate.

@saudet Do you think this is a good default behaviour? If so I will create a PR.

saudet, 2022-02-11:
That sounds like a good idea, yes. The default imageWidth, imageHight, and frameRate should be taken from the properties though: https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/FrameGrabber.java#L183

saudet, 2022-02-15:
Thanks @cansik for fixing this! @hyeonjang please give it a try by replacing `modules/javacv.jar` with this file: https://oss.sonatype.org/content/repositories/snapshots/org/bytedeco/javacv/1.5.8-SNAPSHOT/javacv-1.5.8-20220215.010447-6.jar

hyeonjang, 2022-02-16:
@saudet @cansik Really thank you for updating! Now it works.

closed 2022-02-16 by hyeonjang