bytedeco/procamcalib #27: Crashes on open

eldrickm, 2022-03-07:
Hello,

I'm receiving the following error (full log attached) upon opening the application.
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007fffa4c121f3, pid=33116, tid=17012
#
# JRE version: Java(TM) SE Runtime Environment (17.0.2+8) (build 17.0.2+8-LTS-86)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (17.0.2+8-LTS-86, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, windows-amd64)
# Problematic frame:
# C  [realsense.dll+0x521f3]
```

I'm on Windows 10 with Oracle JDK 17.02

It seems to be an issue with the realsense dll - has this been seen before?

Thanks!
[hs_err_pid33116.log](https://github.com/bytedeco/procamcalib/files/8200912/hs_err_pid33116.log)

saudet, 2022-03-07:
If you don't have a RealSense camera, you can just remove those files from the modules subdirectory.

eldrickm, 2022-03-08:
Hi Samuel,

Unfortunately I am using a RealSense camera. I can confirm that removing `librealsense.jar` does prevent the crash, but I cannot open the RealSense camera device in the application.

Are there other workarounds?

Thank you!

saudet, 2022-03-08:
Well, it's obviously some issue with librealsense itself. Someone will need to debug this...

poqudrof, 2022-03-09:
Hello, I have been tagged here. 

A possible issue would be that the DLL does not match the same API as the Intel SDK. 
I am not sure how the code evolved in the past years, however the java versions were much older and the Intel Realsense SDKs changed a lot. 
I still have a realsense camera, but I am not sure to be able to make it work. 

What camera do you use, and which version of Realsense Drivers ?

eldrickm, 2022-03-10:
Hi Jeremy,

I'm using the L515 (the ToF one) and am using the v2.5.0 of the SDK (https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0)

saudet, 2022-03-10:
If you have a camera that works with version 2.x of the SDK, use "RealSense2" instead. You can ignore "RealSense".