bytedeco/procamcalib #28: Getting severe error on start

MoritzSkowronski, 2022-08-03:
Hi,

on my Windows 11 64 machine with jdk 18 installed, I get the following error on startup:

```
SEVERE: No way to find original stream handler for jar protocol
java.lang.reflect.InaccessibleObjectException: Unable to make field transient java.net.URLStreamHandler java.net.URL.handler accessible: module java.base does not "opens java.net" to unnamed module @40f9161a
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
```

Am I using the wrong idk?

saudet, 2022-08-03:
Right, it looks like we need JDK 8 or 11.

MoritzSkowronski, 2022-08-03:
Wow, that fast! ;) 

Nooby question: Is there any way to download idk 8 or 11 _without_ having to register on the oracle site?!

saudet, 2022-08-03:
Yes, it's all available from here: https://adoptopenjdk.net/

saudet, 2022-11-02:
BTW, it's also possible to use JDK 17, or more recent, by enabling a couple of parameters that I've added to commit https://github.com/bytedeco/procamcalib/commit/e0e853e0b708459972953c0204e833c4deea33e7. Those will be needed until NetBeans adds support for newer versions of the JDK.

MoritzSkowronski, 2022-11-03:
Thanks for the update! Will check it out asap :)