bytedeco/procamcalib #29: Error when using 1024 x 768 projector resolution

stiefen1, 2022-08-10:
Hello,

I'm trying to calibrate a projector-camera system (Azure Kinect + BenQ MX631ST) on Ubuntu 20.04. If I use the default projector resolution (1080 x 1920) the program runs normally (images are correctly received from the camera and the pattern is projected through the projector) but if I change these values to 768 x 1024, which is the resolution of my projector, I get the following error when I press the start button : 

[image]

Is there any other parameters I should change such as a ratio ? Here is the configuration of the projector I use when the error appears:

[image]

Thank you very much for your help!

saudet, 2022-08-10:
It sounds more like an issue with the JDK. Could you try again with JDK 8 or 11?

stiefen1, 2022-08-10:
Thank you for your very quick answer! 
Actually I was already on open JDK 11 and tried to downgrade to open JDK 8 but I still have the same problem. Maybe I forgot to install something, I'm not use to Java. The complete error message is the following:

openjdk version "1.8.0_342"
OpenJDK Runtime Environment (build 1.8.0_342-8u342-b07-0ubuntu1~20.04-b07)
OpenJDK 64-Bit Server VM (build 25.342-b07, mixed mode)
RealSense devices found: 0
RealSense2 devices found: 0
(1) Could not find the CLEyeMulticam.dll
(2) Could not find the CLEyeMulticam.dll
Aug 10, 2022 2:14:27 PM org.openide.util.ImageUtilities$CachedLookupLoader getLoader
WARNING: No interface org.openide.util.spi.SVGLoader instance found in ProxyLookup(class=class org.netbeans.core.startup.MainLookup)->[MetaInfServicesLookup[sun.misc.Launcher$AppClassLoader@677327b6], SingletonLookup[sun.misc.Launcher$AppClassLoader@677327b6], org.openide.util.Lookup$Empty@57f49b3, AbstractLookup[]]
Aug 10, 2022 2:14:27 PM org.openide.util.ImageUtilities getIcon
INFO: No SVG loader available for loading jar:file:/home/smonnet/calibration/procamcalib-1.5.7-bin/modules/org-netbeans-swing-plaf.jar!/org/netbeans/swing/plaf/resources/hidpi-folder-closed.svg
Aug 10, 2022 2:14:27 PM org.openide.util.ImageUtilities getIcon
INFO: No SVG loader available for loading jar:file:/home/smonnet/calibration/procamcalib-1.5.7-bin/modules/org-netbeans-swing-plaf.jar!/org/netbeans/swing/plaf/resources/hidpi-folder-open.svg
org.bytedeco.javacv.FrameGrabber$Exception: Failed to load class org.bytedeco.javacv.FlyCapture2FrameGrabber
	at org.bytedeco.javacv.FlyCapture2FrameGrabber.tryLoad(FlyCapture2FrameGrabber.java:123)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.bytedeco.javacv.FrameGrabber.init(FrameGrabber.java:54)
	at org.bytedeco.procamcalib.MainFrame.main(MainFrame.java:1280)
Caused by: java.lang.UnsatisfiedLinkError: no jniFlyCapture2 in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1860)
	at java.lang.Runtime.loadLibrary0(Runtime.java:871)
	at java.lang.System.loadLibrary(System.java:1124)
	at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:1800)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1402)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1214)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1190)
	at org.bytedeco.flycapture.global.FlyCapture2.<clinit>(FlyCapture2.java:14)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1269)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1214)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1206)
	at org.bytedeco.javacv.FlyCapture2FrameGrabber.tryLoad(FlyCapture2FrameGrabber.java:121)
	... 6 more
Caused by: java.lang.UnsatisfiedLinkError: /home/smonnet/.javacpp/cache/flycapture-linux-x86_64.jar/org/bytedeco/flycapture/linux-x86_64/libjniFlyCapture2.so: libflycapturevideo.so.2: cannot open shared object file: No such file or directory
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1934)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1817)
	at java.lang.Runtime.load0(Runtime.java:810)
	at java.lang.System.load(System.java:1088)
	at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:1747)
	... 16 more
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found 0 devices
RealSense2 devices found: 0
Aug 10, 2022 2:15:01 PM org.bytedeco.procamcalib.MainFrame calibrationStartMenuItemActionPerformed
SEVERE: Could not initialize calibration worker thread.
org.bytedeco.javacv.CanvasFrame$Exception: Failed to create CanvasFrame
	at org.bytedeco.javacv.ProjectorDevice.createCanvasFrame(ProjectorDevice.java:328)
	at org.bytedeco.procamcalib.CalibrationWorker.init(CalibrationWorker.java:172)
	at org.bytedeco.procamcalib.MainFrame.calibrationStartMenuItemActionPerformed(MainFrame.java:981)
	at org.bytedeco.procamcalib.MainFrame.calibrationStartButtonActionPerformed(MainFrame.java:1225)
	at org.bytedeco.procamcalib.MainFrame.access$300(MainFrame.java:121)
	at org.bytedeco.procamcalib.MainFrame$6.actionPerformed(MainFrame.java:583)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
[... 18 lines cut ...]
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
[... cut ...]
	at org.bytedeco.javacv.CanvasFrame.<init>(CanvasFrame.java:141)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.bytedeco.javacv.ProjectorDevice.createCanvasFrame(ProjectorDevice.java:310)
	... 43 more

Do you have any other idea ? Thank you very much again.

saudet, 2022-08-10:
> Caused by: java.lang.UnsupportedOperationException: Cannot change display mode

Ah, here's the problem. It looks like you'll need to change the display resolution to 1024x768.

stiefen1, 2022-08-10:
Thank you very much this solved my problem! And thank you also for your quick and accurate answers!

closed 2022-08-10 by stiefen1