bytedeco/procamcalib #30: procamcalib doesn't run(doesn't even show the UI) when calibration one camera and a projector

Michael2333, 2023-02-26:
I try to calibrate a single camera(BASLER) with a projector but calibration doesn't run，doesn't even show the UI.
I am using windows 7 and java 1.8.
When i run the command procamcalib-nativelook.cmd, I get as following :

C:\Users\XXXX\Desktop\procamcalib-bin>procamcalib-nativelook.cmd
java version "1.8.0_0271"
Java(TM) SE Runtime Environment (build 1.8.0_271-b09)
Java HotSpot(TM) 64-Bit Server VM (build 25.271-b09, mixed mode)

Then no UI is shown and nothing else.
I only installed jdk8, do I need to install something else

help please，thanks!

saudet, 2023-02-26:
Windows 7 might not work, it's not being tested. Please try with Window 10.

Michael2333, 2023-02-27:
Got it!   thanks for the quick reply!!  i'll try Window 10.

codespearhead, 2024-03-25:
@Michael2333 Did it work? If so, can you close this issue?