bytedeco/procamcalib #31: Unable to run on Apple Silicon Mac

zumpchke, 2024-03-23:
```
Last login: Sat Mar 23 00:52:28 on ttys002
vanush@vanbook ~ % java --version
vanush@vanbook ~ % cd ~/repos/procamcalib 
vanush@vanbook procamcalib % cd ~/Downloads/
vanush@vanbook procamcalib % clear

vanush@vanbook procamcalib % cd ~/Downloads/procamcalib-1.5.9-bin 
vanush@vanbook procamcalib-1.5.9-bin % ./procamcalib-oceanlook 
Unrecognized option: -d64
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
WARNING: 64-bit server JVM not available. Executing with default JVM...
(1) Could not find the CLEyeMulticam.dll
(2) Could not find the CLEyeMulticam.dll
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openide.util.RequestProcessor$TopLevelThreadGroup (file:/Users/vanush/Downloads/procamcalib-1.5.9-bin/modules/org-openide-util.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of org.openide.util.RequestProcessor$TopLevelThreadGroup
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
org.bytedeco.javacv.FrameGrabber$Exception: Failed to load class org.bytedeco.javacv.FlyCapture2FrameGrabber
	at org.bytedeco.javacv.FlyCapture2FrameGrabber.tryLoad(FlyCapture2FrameGrabber.java:123)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.bytedeco.javacv.FrameGrabber.init(FrameGrabber.java:55)
	at org.bytedeco.procamcalib.MainFrame.main(MainFrame.java:1280)
Caused by: java.lang.UnsatisfiedLinkError: Platform "macosx-arm64" not supported by class org.bytedeco.flycapture.global.FlyCapture2
	at org.bytedeco.javacpp.Loader.load(Loader.java:1261)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1234)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1226)
	at org.bytedeco.javacv.FlyCapture2FrameGrabber.tryLoad(FlyCapture2FrameGrabber.java:121)
	... 6 more
org.bytedeco.javacv.FrameGrabber$Exception: Failed to load class org.bytedeco.javacv.OpenKinect2FrameGrabber
	at org.bytedeco.javacv.OpenKinect2FrameGrabber.tryLoad(OpenKinect2FrameGrabber.java:105)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.bytedeco.javacv.FrameGrabber.init(FrameGrabber.java:55)
	at org.bytedeco.procamcalib.MainFrame.main(MainFrame.java:1280)
Caused by: java.lang.UnsatisfiedLinkError: Platform "macosx-arm64" not supported by class org.bytedeco.libfreenect2.global.freenect2
	at org.bytedeco.javacpp.Loader.load(Loader.java:1261)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1234)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1226)
	at org.bytedeco.javacv.OpenKinect2FrameGrabber.tryLoad(OpenKinect2FrameGrabber.java:100)
	... 6 more
org.bytedeco.javacv.FrameGrabber$Exception: Failed to load class org.bytedeco.javacv.RealSenseFrameGrabber
	at org.bytedeco.javacv.RealSenseFrameGrabber.tryLoad(RealSenseFrameGrabber.java:106)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.bytedeco.javacv.FrameGrabber.init(FrameGrabber.java:55)
	at org.bytedeco.procamcalib.MainFrame.main(MainFrame.java:1280)
Caused by: java.lang.UnsatisfiedLinkError: no jniRealSense in java.library.path: [/Users/vanush/Library/Java/Extensions, /Library/Java/Extensions, /Network/Library/Java/Extensions, /System/Library/Java/Extensions, /usr/lib/java, .]
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2678)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:830)
	at java.base/java.lang.System.loadLibrary(System.java:1886)
	at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:1832)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1423)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1234)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1210)
	at org.bytedeco.librealsense.global.RealSense.<clinit>(RealSense.java:14)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:398)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1289)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1234)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1226)
[... 56 lines cut ...]
	at org.bytedeco.javacpp.Loader.load(Loader.java:1234)
	at org.bytedeco.javacpp.Loader.load(Loader.java:1210)
	at org.bytedeco.artoolkitplus.global.ARToolKitPlus.<clinit>(ARToolKitPlus.java:14)
	at org.bytedeco.javacv.Marker.getImage(Marker.java:104)
	at org.bytedeco.javacv.Marker.getImage(Marker.java:97)
	at org.bytedeco.javacv.Marker.draw(Marker.java:126)
	at org.bytedeco.javacv.Marker.draw(Marker.java:114)
	at org.bytedeco.javacv.MarkedPlane.setPrewarp(MarkedPlane.java:149)
	at org.bytedeco.javacv.MarkedPlane.<init>(MarkedPlane.java:95)
	at org.bytedeco.javacv.MarkedPlane.<init>(MarkedPlane.java:39)
	at org.bytedeco.procamcalib.MainFrame.updatePatterns(MainFrame.java:310)
	at org.bytedeco.procamcalib.MainFrame.loadSettings(MainFrame.java:447)
	at org.bytedeco.procamcalib.MainFrame.<init>(MainFrame.java:206)
	... 14 more
Caused by: java.lang.UnsatisfiedLinkError: Could not find jniARToolKitPlus in class, module, and library paths.
	at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:1799)
	... 28 more

^C%                                                                             vanush@vanbook procamcalib-1.5.9-bin % org.bytedeco.javacv.Marker.getImagejc
vanush@vanbook procamcalib-1.5.9-bin % java --version
openjdk 11.0.21 2023-10-17 LTS
OpenJDK Runtime Environment Zulu11.68+17-CA (build 11.0.21+9-LTS)
OpenJDK 64-Bit Server VM Zulu11.68+17-CA (build 11.0.21+9-LTS, mixed mode)
vanush@vanbook procamcalib-1.5.9-bin % 
```

saudet, 2024-03-23:
Duplicate of https://github.com/bytedeco/javacpp-presets/issues/1069

closed 2024-03-23 by saudet

zumpchke, 2024-03-23:
@saudet M1 can run x86 code .. is there a workarund?

saudet, 2024-03-23:
If you use an x86 version of the JDK it will work, yes.

zumpchke, 2024-03-23:
I havent been able to get an x86 version to run correctly.

Is it possible to build everything from scratch locally? I'm from c/c++ and can't figure out Maven. Any hints?

saudet, 2024-03-23:
Sure, please follow the instructions here: 
https://github.com/bytedeco/javacpp-presets?tab=readme-ov-file#build-instructions